cmcgerty/Sofar2mqtt #1: Rs485 to ttl with auto flow control

cliffdude, 2021-01-23:
Hi,

Would this code work on an auto flow rs485 to ttl module the one that does not have RE and DE, only rx, tx and vcc and gnd? If not could it be modified to work?
I want to try it on a HYD-4000-ES model but I only have the auto rs485-ttl converter on hand.

cmcgerty, 2021-01-23:
Hi,

Yes, it is tested on this one...
https://www.amazon.co.uk/gp/product/B08HQMDKGR/ref=ppx_yo_dt_b_asin_title_o09_s00?ie=UTF8&psc=1

I've not tested the modbus instructions on anything but the ME3000SP but the Sofar modbus documentation is not model specific. Certainly the Charge and Discharge modes will not exceed 3000W at the moment, but you can change lines 410 and 427 to allow for more. 

If you start by testing the read only functions and don't go into Passive Mode, you can see if the output looks sensible.

Let me know how it goes.

cliffdude, 2021-01-24:
UPDATE:
So I built the module with a D1 mini and used the auto flow converter...  and it works!! Awesome work my friend you rock 🥇 
In case anyone is madly googling a solution for Sofar Inverter to MQTT I can confirm this is working on the HYD 3-6000 ES models. I have not tested the battery mode yet but I am getting all values via wifi on my MQTT server. Now to build my home assistant dashbord.
RS485 to TTL used was:
https://ae01.alicdn.com/kf/H5299716877f943ae80442061597977df9.jpg 
Just to clarify for others, auto flow on uses RS, TX and GND so for this project wemos pin D6 to RX and D7 to TX GND and 5V are joined on Wemos and converter module.
Thank you @cmcgerty 👍

closed 2021-01-24 by cliffdude

cmcgerty, 2021-01-24:
It’s great to hear that it can read the data for the HYD models. Thanks for the update. I’ll add to the Readme. Let me know how it goes with controlling the charge mode.

I built the first prototype with an auto flow type TTL board so I was knew it had some flexibility there. Glad it works with the one you have.

megamarco833, 2022-12-28:
> UPDATE: So I built the module with a D1 mini and used the auto flow converter... and it works!! Awesome work my friend you rock 🥇 In case anyone is madly googling a solution for Sofar Inverter to [...]

hello i have the same TTL converter but i see some issue:
https://github.com/cmcgerty/Sofar2mqtt/issues/51#issue-1513037724

maybe can you explain how you connect A+ and B- to your inverter?
you used also the GND (beside the B-) ?
thanks