cmcgerty/Sofar2mqtt #7: battery_save doesnt stop discharging in auto mode

graememk, 2021-03-21:
Looking at the code, setting battery_save active does not trigger the inverter to change modes.

Possible solution, when battery_save is set active, Put the inverter into standby, then the battery_save routine can take over.

cmcgerty, 2021-03-21:
Hi,

The subroutine batterySave() runs on a timer, once every three seconds. It checks the BATTERYSAVE flag and if true will switch between auto or standby depending if there is excess solar exergy. If energy is flowing to the grid, it switches to auto mode in order to allow the battery to charge, if energy is being imported from the grid then there is no excess solar and the inverter switches to standby. So line 451 only needs to set BATTERYSAVE true, the rest is handed by the already running timer.

graememk, 2021-03-21:
Ah I understand,

What if the inverter is already in auto, Discharging, Nothing is being imported from grid, battery is covering the current usage.

Then the off peak time happens, and I want to switch it to battery_save  , sending the battery_save command doesn't stop the inverter from discharging.  

But if I send the standby command first and then the battery save, the standby stops the discharge and then the battery save does its business.

graememk, 2021-03-22:
Just after I finished typing that, I had the ME3000 discharging to see the behaviour, and my EV charge kicked in, and that brought the Discharge to a stop. As you described.

closed 2021-03-22 by cmcgerty

graememk, 2021-03-22:
There is still an edge case here, as it's monitoring the grid power, 

My scenario was that the inverter was keeping the grid power at 0, from batteries not solar. Only when a high load kicked (say EV, kettle etc) in that it exceeded the output of the me3000. 

That's why I suggested initially setting it to standby to get the grid power value not 0. Then the battery save routine can take over and change between standby and auto as needed.

Monitoring the battery discharge could be an option too.

Needless to say I do love the project, have built 3 of them so far for myself and a few friends.