cmcgerty/Sofar2mqtt #9: MQTT Rarely Updates

Sku1ly, 2021-08-04:
Hi,

Thanks for making this, I have everything set up and working but I get updates via MQTT maybe every ten minutes and usually only six times. Reboot the nodemcu sometimes brings it back, sometimes doesn't.

Serial monitor shows constant MQTT reconnects and I can see the same in the MQTT server logs.

How often is it supposed to update and have you seen similar behaviour before?

Sku1ly, 2021-08-04:
I think I may have solved this. I think I chose the wrong nodemcu version 1.0 vs 0.9 which caused frequent WiFi drop outs.

Seems to be stable now so will close.

Sku1ly, 2021-08-05:
It wasn't this :(
I've tried two nodemcus and a wemos d1 mini and seeing the same behaviour.

Pinging I see constant drops and errors in serial monitor. It worked for most of today then broke.

Any thoughts?

[image]

zambujal, 2021-09-30:
Don't know... but it's weird... could be some wire, sometimes things like this happens to me... I redo everything... and then it's working. ;)

cmcgerty, 2021-09-30:
Out of interest, did you also change the ESP8266? There is some question mark in my mind as to whether some are more tolerant of 3.3v logic than others. That said, I normally just use the cheapest ones I can find and don’t have an issue.

zambujal, 2021-10-01:
I used the lol nonemcu cheap 8266 it worked well for me.

cmcgerty, 2022-04-24:
Hi @zambujal .

Sofar2mqtt V2.0 beta1 has been published today, which is a major write and should bring much greater stability. See if that is better for you.