cmcgerty/Sofar2mqtt #10: MQTT random update value

HeartandHana, 2021-08-14:
Thank you for discussing this in advance. I found a problem when received MQTT public msg. Some public msg. was show some value such as the first time show the inverter status. Sometimes don't show it.  Please see the example payload below. I would like to need all value to the public all time.  Please help me. (for information I used inverter model hyd5500 in Thailand).  Appreciated again.
[image]

cmcgerty, 2022-04-24:
Hi @HeartandHana .

Sofar2mqtt V2.0 beta1 has been published today, which is a major write and should bring much greater stability. See if that is better for you. I'm especially keen to get feedback from owners of hybrid inverter models as we have now combined it into a single file. Just uncomment your model at the top of the file.

cliffdude, 2022-04-24:
I will try it asap. HYD-4000-ES.

cmcgerty, 2022-04-27:
Thanks @cliffdude 

Hopefully it will work well for HYD models. Let me know how you get on.