cmcgerty/Sofar2mqtt #13: Json from ESP local server

jezikk82, 2021-11-19:
Hi, would it be possible to present data in json format on local (on ESP) Web server? 
I want to integrate it with smart home solution based on plc.

I don't have mqtt server and gateway for my plc so it would be much easier to work with json or any other format with i can get by http request from.

cliffdude, 2021-11-21:
Check out my git SofarCtrl it is a modified copy of Colin's code with a webserver and OTA support.
All original code is there but I have comented out what I did not need such as OLED support.
https://github.com/cliffdude/SofarCtrl/blob/master/HYBRID-WEB_OTA.ino

schritta, 2021-11-30:
@cliffdude 
When using the Hybrid-web_OTA , do we still need the mqtt server?
I only need the readings on the local web(esp) and then do a http request from node-red.
Is it then correct not using the mqtt stuff in you code?

jezikk82, 2021-11-30:
Yes, you can not use mqtt. I just commented it out and rest "works".

cliffdude, 2021-11-30:
Yes, you are correct, I use personally mqtt but you do not need to! Check in the code and you will understand what to do.

schritta, 2021-11-30:
Thank you for the quick response.  I have ordered the parts as soon as the arrive I will start this project.
And keep you posted. I aim looking into the code to see what exactly to comment out.

schritta, 2021-11-30:
@jezikk82 
Where did you make the connection for the RS485 in your HYBRID, because the pics show only the ME3000SP.
Is there also a 485s connector? I have got the HYB 6000-ES.

jezikk82, 2021-11-30:
[image]

First from the left is modbus for battery, second (green plug) is for slave devices and you connect max485 with ESP there.

schritta, 2021-11-30:
@jezikk82  Thank you very much i'll check this also with my device.

schritta, 2021-12-04:
Ok , received the espc and setup the hardware on temporary project board.
It works for 99%. Where did you got the register information? Because I don't get "Solarpv2", all other readings are received . So I thought there is a minor issue with the reading of that specific register. I do not know how to debug this.

schritta, 2021-12-04:
my uploaded code in esp
[Hybrid-web_OTA.txt](https://github.com/cmcgerty/Sofar2mqtt/files/7654007/Hybrid-web_OTA.txt)

cliffdude, 2021-12-04:
I think this should be what you need. If you can take the time to debug it
that would be great as I sometime get strange values that I need to filter
out via Home Assistant in order to not mess up my graphs and energy
calculations.

https://docplayer.net/200949628-Sofarsolar-modbus-rtu-communication-protocol.html

schritta, 2021-12-04:
thanks I'll try to debug it, but I aim not a specialist.

jezikk82, 2021-12-04:
here you have newer version of modbus for hybrid inverters. 
[SofarHYD ES ME3000SP Modbus protocol.pdf](https://github.com/cmcgerty/Sofar2mqtt/files/7654366/SofarHYD.ES.ME3000SP.Modbus.protocol.pdf)

What inverter and datalogger firmware  you have ?
I have v302 and LSW3_15_FFFF_1.0.78 and in my case I cant get any data neither by ModBus RTU or TCP :(

cliffdude, 2021-12-04:
I have the LSW3_14_FFFF_1.0.00 and a HYD-4000-ES with Pylontech Li-on
batteries.
The Wifi logger has nothing to do with this ESP program as it gets data via
the RS485 port on the inverter itself.
You should see an rs485 logo on the inverter screen at the top left if you
have correctly connected the adapter. If you adapter has Leds they should
also flash both tx and rx.
Try swapping the rs485 + and - wires on the inverter.
I did try to connect to the inverter with my pc and an adapter but that did
not work!!!

Good luck

schritta, 2021-12-04:
I have  SW version 2.90 HW version 1.00 HYB-5000-ES, and yes i got the rs485 symbol in the left upper corner .
@jezikk82 try indeed to swap the + and - rs485 wires.

jezikk82, 2021-12-04:
I see the rs485 logo but the when I try to get data from inverter I got crc error on all 0x03 function code and all registry. 
I can take info from 0x04 function and 0x2000 registry = the are serial no, firmware ver etc.

richplatts, 2021-12-04:
The registry is slightly different for HYD
I need to get my windows laptop out to send you my version

Richard Platts  -  07863 249063

schritta, 2021-12-17:
I have received the mod-bus RTU protocol from sofar 5 days ago, but the Solar PV2 has the same modbus address so i did a calculated solarPv2 value in node red "SolarPV - SolarPV1 = SolarPV2" so now I've got all the data needed to build my node red flows. The most important addresses are listed on page 15 Panel string information see pdf.
Now everything works as planned for me. Thank you to remind me for updating.
[SofarHYD ES Modbus protocol.pdf](https://github.com/cmcgerty/Sofar2mqtt/files/7738102/SofarHYD.ES.Modbus.protocol.pdf)

cmcgerty, 2022-04-24:
Hi All.

Sofar2mqtt V2.0 beta1 has been published today, which is a major write and should bring much greater stability. Any testing would be greatly appreciated. I'm especially keen to get feedback from owners of hybrid inverter models as we have now combined it into a single file. Just uncomment your model at the top of the file.

@cliffdude I still really like your fork that delivers the json output directly. For a Node-Red user like me I could skip the mqtt server entirely. Sorry if it's too much work to merge them. But it might be something we can collaborate on.

cliffdude, 2022-04-24:
I will try your  V2 as I gave stability issues with some of the returned
values Most of the it's fine but I get the odd values that have to be
filtered out. It could be my hardware... If it works better I will try to
add the webserver component and Json output to it.
I just customized your original for my needs.

cliffdude, 2022-04-24:
> Hi All.
>
> Sofar2mqtt V2.0 beta1 has been published today, which is a major write and should bring much greater stability. Any testing would be greatly appreciated. I'm especially keen to get feedback [...]

Would be happy to merge but I removed the LCD that I did not need and we should merge for V2.