cmcgerty/Sofar2mqtt #14: No data on Sofar Hyd-3000-es

jezikk82, 2021-11-20:
Hi,

I try to get it working on Sofar hybrid inverter. I checked the addresses and they are correct for my inverter, but unfortunately all the time i get the following error in console and no data are returned:
Send heartbeat
Error: invalid data frame

In function updateRunstate(); when I  print 
Serial.println(responce.data[0]); => 218
Serial.println(responce.data[1]); => 255

Can you help me ?

jamesworts, 2022-01-21:
I also have this issue, any solutions?

jezikk82, 2022-01-21:
Unfortunately no, I tested 4 different firmwares (now I'm at v302) and on all the same problem.

Sofar is blind on that problem, I found couple more people with the same issue. Try to send them mails on different addresses, maybe they will react when more people will highlight it.

cliffdude, 2022-01-21:
Sorry it's not very helpful but I have v285 on the HYD-4000-ES and it works
fine. Sounds like Sofar may have changed something in the v3 firmwares.

jezikk82, 2022-01-21:
I started with 235 from facrory, later tested 270, 280, 300, 302. On none of this firmwares it worked.

1. Can you share your firmware version?
2. What version of data logger (lsw) you have?
3. Do you think firmware from hyd400 will work on hyd3000?

jamesworts, 2022-01-22:
mine is v310 and not working.

cliffdude, 2022-01-22:
Hi,

The wifi adapter has nothing to do with Colin's device / code, as this uses
the RS485 port that is for talking to a paired inverter in a Master / Slave
configuration.
The HYD-xxxx-ES series are identical (as far as I know), I have friend with
a HYD-5000-ES and it is the same, he is running v280, I would even go as
far as to think that the output limitation is in the software?!?
Are you sure that your TTL adapter is ok, I built 3 of them and only one
works? Aliexpress... what can I say. It is far more likely to be an issue
with your device / build than a firmware issue with the inverter, also
check your power source for the esp and ttl - rs485.

cmcgerty, 2022-04-24:
Hi All.

Sofar2mqtt V2.0 beta1 has been published today, which is a major write and should bring much greater stability. See if that is better for you. I'm especially keen to get feedback from owners of hybrid inverter models as we have now combined it into a single file. Just uncomment your model at the top of the file.

innerlurken, 2022-04-28:
Hi, I'm using a HYD5000 hybrid inverter and have updated to version 2, it's working much better, seems very stable now, thank you

Question, do you have an updated node red function? The current one does not seem to work for me.

davies8172, 2022-04-28:
> Hi, I'm using a HYD5000 hybrid inverter and have updated to version 2, it's working much better, seems very stable now, thank you
> 
> Question, do you have an updated node red function? The current one does not seem to work for me.

If you are looking for up to date Node-Red flows, goto https://www.facebook.com/groups/2477195449252168
You will find the most recent Node-Red flow (for Sofar2MQTT) at https://www.facebook.com/groups/2477195449252168/permalink/2570410946597284

innerlurken, 2022-04-30:
Thank you, the new flow helped loads, the new version is very good, thank you for all your work on this