cmcgerty/Sofar2mqtt #16: crash in discharge mode

Argenlos, 2022-02-11:
Hello, and thank you for this useful tool.
I have modified your code a bit to display in the console the results, because I don't have an OLED, but I have my data displayed in the console..everything is fine.
But, I have a problem.
When the ME3000SP inverter goes into "discharge" mode by itself, at the end of the day, you hear the inverter switches, to switch to discharge mode, the esp8266 crashes and restarts all the time.

In charge mode and in standby no worries.

I really don't know where to start looking for the problem, because this little assembly works...but not in discharge.

The firmware is V280 on the inverter
I use a nodemcu rather than a D1 mini. 
I tested with another power supply just in case.

[error.txt](https://github.com/cmcgerty/Sofar2mqtt/files/8051275/error.txt)
Good evening.

cmcgerty, 2022-02-11:
Sorry to say that I have seen this problem before with devices other than esp8266. Even cheap esp8266s can have this problem. I think the issue is how tolerant it is of the MAX485 board, which would prefer a 5v input. If you use a MAX3485 it may be more stable. And the wiring is actually simpler as MAX3485 doesn't need the DE/RE wire as it is auto-switching. The only reason I don't recommend them to everyone is they are much less common.

Thanks,
Colin

Argenlos, 2022-02-12:
Hello, thank you for your quick feedback.
I have just tested in 5V on the MAX485 with another ESP8266 and another MAX485.
This morning, when charging the battery, the frames are no longer recognized when switching from standby mode to charging.
I will follow your advice and think about switching to a MAX3485 and another eps8266 of better quality.
I will keep you informed of progress.
Have a good day

Argenlos, 2022-02-17:
Hello,

I received 2 new D1 mini. One clone and one original (Lolin V3.1.0) that I test with a MAX485...and well it's worse, no return even in standby mode on the inverter, the D1 connects the wifi, the mqqt, then crashes with hexa code pages. It reboots again and again.
I put an external power supply of 5V (5A) for the D1 and another for the MAX485...you never know, but it's not that.

I redid the board completely...maybe...but it's not that...(no choice because I had to redo the board for a D1 mini)

I just took the INO file, added the libraries (in Arduino IDE), added the Wifi and Mqtt parameters, compiled without worries (with the right type of board)...and it's not the first assembly with esp that I do.

Of course, I tested with a second, a third MAX485 (component tolerance, etc...), still not that.

With an MCU node at least in standby mode it gives consistent values on frequency, voltage ... (with a very dirty Serial.print in the code), with a D1 mini clone or an original D1 mini, and although frame errors. 

Now let's find a MAX3485...:)

Bye

Argenlos, 2022-02-20:
Hello,

The MAX3485 adapter saved me :)
I think you should recommend the MAX3485 before trying the MX385 which doesn't do the job, or with a lot of luck.

Thanks for the advice.

Sincerely,
[image]

closed 2022-02-20 by Argenlos