cmcgerty/Sofar2mqtt #21: Support for HYD 8KTL-3PH

pixellos, 2022-05-07:
Hi, wanted to ask if given inverter is supported.

cmcgerty, 2022-05-07:
Hi,

I would expect this model to be supported. Use version 2.0, uncomment line 3 and comment out line 2.

Please update this thread if you are able to confirm it working on this model.

Thanks

cliffdude, 2022-05-08:
If you can confirm that the protocol is the same as the other HYD models,
and I don't see why it would not be, then it should work. I have a single
phase HYD model and it works fine.
The only way would be to ask for the protocol details on the forum or
directly from Sofar, the problem is they are not always willing to give out
that information.

maxxximatoze, 2022-05-11:
> If you can confirm that the protocol is the same as the other HYD models, and I don't see why it would not be, then it should work. I have a single phase HYD model and it works fine. The only way [...]

Hello, what model of HYD single phase do you have and how do you make it works with it ? Because I have a HYD6000-EP, and I can't make it works with it for now (:

cmcgerty, 2022-10-25:
Not very promising. Have you tried swapping the two RS485 wires?

tomipaci, 2022-10-25:
Yes, I think it then reports an RS485 error. Do you have any idea? Or does the HYD 15 KTL have a different addressing?

ruben05041991, 2023-01-15:
Anyone who got it working for a HYD xx KTL 3PH inverter?

LotharF, 2023-02-07:
I'm also use a HYD 8KTL 3PH and try to use the latest Sofar2mqtt Version 2.1. But it will not work.
I connect the A and B cable with Pin 1+4 of the "Multiconnector" of my HYD. The only result is "Heartbeat -2".

pixellos, 2023-02-07:
Maybe U want to try a project i hacked some time ago - https://github.com/pixellos/codereinvented.automation.py - its communicating thru wifi logger, so no custom board etc

Mocky158, 2023-03-14:
I would also like hyd KTL support I have version 20
I have an excel document with the given registers for the HYD KTL version if that helps.
[SOFAR-G3 External Modbus Protocol-EN-V1.10 (20220622).xlsx](https://github.com/cmcgerty/Sofar2mqtt/files/10964325/SOFAR-G3.External.Modbus.Protocol-EN-V1.10.20220622.xlsx)

hruskaexe, 2023-09-13:
Hi, please will this issue be resolved, is it being worked on? Thanks

Stronzo-sky, 2023-11-02:
hi, i have hyd-5k20ktl-3ph - any chance theese invertors will be supported with Sofar2mqtt?

wichr, 2024-05-14:
This version suport Sofar HYD KTL. https://github.com/IgorYbema/Sofar2mqtt

LotharF, 2024-05-15:
Shure? I have installed the software and get only mqtt values with "0" via mqtt. I use the "HYBRID.ino". Maybe I have to change something more here if I have a SofarSolar HYD 8KTL 3PH ?

LotharF, 2024-07-19:
I have some trouble with the version from IgorYbema. It was running for a while and then no data. After I change the rs485 board all is running fine again. And nor the same. Tee re485 board seems to be damaged again. Is this normal or is the rs485 board from Amazon to cheap to run for long time?

cliffdude, 2024-07-20:
I had the same issue, I had to restart the ESP periodically. In the end it
was the power supply that I was using that was also connected to my
pylontech battery.

peterdedecker83, 2025-02-20:
On my ME3000SP I have the same problem, already tried swapping the two wires. What else could be wrong and how to investigate? Thanks.

21:45:50.724 -> Response too short
21:45:51.304 -> Timeout waiting for RS485 response.
21:45:51.337 -> Response too short
21:45:51.919 -> Timeout waiting for RS485 response.
21:45:51.952 -> Response too short
21:45:52.533 -> Timeout waiting for RS485 response.
21:45:52.564 -> Response too short
21:45:52.693 -> Send heartbeat
21:45:53.309 -> Timeout waiting for RS485 response.
21:45:53.341 -> Response too short
21:45:53.341 -> Bad heartbeat -2
21:45:53.372 -> Get runstate: Timeout waiting for RS485 response.
21:45:54.016 -> Response too short
21:45:54.016 -> Response too short
21:45:54.661 -> Timeout waiting for RS485 response.
21:45:54.694 -> Response too short
21:45:55.274 -> Timeout waiting for RS485 response.
21:45:55.274 -> Response too short
21:45:55.859 -> Timeout waiting for RS485 response.
21:45:55.891 -> Response too short
21:45:56.475 -> Timeout waiting for RS485 response.
21:45:56.508 -> Response too short