cmcgerty/Sofar2mqtt #27: How do you power the ESP8266?

jaimelaborda, 2022-07-05:
I have a HYD-4600-ES, and couldn't found a 5V pin to power the microcontroller from the inverter.

[image]

Any of you with the same problem? How have you solved it?
I would like to try as much as possible not to have to power it externally.

Thank you!

Ernest-burgess, 2022-12-03:
hi do you not have a port for conecting the wifi interface that they use to send data to the cloud
this will have a 5 volt output i would expect on this interface to power the external dongal/ wifi transmitter .
i have also fitted a small 5 volt power suppy in the case with two red mains wires  and two 5 vdc wires these can run up to 1 amp at 5 volts and these seem to work well been running them for over 2 years  it is a ebay item 253302424410  these are in there own case so mains is isolated from your board you need to extend the two red mains wires and include a mains fuse only needs to be small 500ma to 1Amp  then you can conect this to the mains input terminal of the inverter.

headcloudmonkey, 2023-03-17:
I've built a small DB9 male/female passthrough to put between the logger and the inverter. Off of that, I pull 5v from pin 9, and ground from pin 5. These are connected to the 5v/gnd pins on the esp8266. Saves running separate supplies for it,.

cliffdude, 2023-03-17:
Great idea, why didn't I think of that? Duh. I have a bunch of those lying
around from old PC serial port connectors.
Have you been running your solution for some time?  I just wondered how
stable the power was with both esp and wifi dongle.

headcloudmonkey, 2023-03-18:
I've had mine using that for a long time and it seems stable for both the ESP8266 and the logger.

[image]

OGB1952, 2023-03-21:
Great tip about getting 5V from the DB9.  Presumably that DB9 also has either RS485 or perhaps TTL level RS232, to send data to the Solarman logger.  I'm keen to get rid of the Solarman logger (don't like having my data going to a server in China).  It's occurred to me that it might be possible to make a replacement for the Solarman stick that includes an ESP8266 running Sofar2mqtt.

It would be a very neat solution to have a new dongle that just plugs into the DB9 and only sends data locally to HA.

Anyone happen to know the pinout for that DB9 connector on the ME3000SP (probably the same on some other Sofar inverters)?

I could reverse engineer the Solarman logger, but given the price of the things I'm reluctant to break it open!

headcloudmonkey, 2023-03-25:
Checked the standard pin numbering for DB9 connectors. On this photo, the pin with the square pad is pin 1, pin 5 ground, pin 9 +5vdc, and 2 & 3 data
[image]

OGB1952, 2023-03-25:
Brilliant!  Many thanks. 

A replacement dongle, with an ESP8266 and perhaps a modified version of Sofar2mqtt could  be very neat and tidy.  It'd be a lot easier to just plug a DB9 into the inverter than faff around with dangling wires and a separate 5V power supply.

Pity there isn't more documentation about these inverters.  Make life a lot easier if they just published the interface spec.

headcloudmonkey, 2023-03-25:
I'm suspecting that it is going to be a read-only interface, given that it's intended to be a logger connected to it, and that they provide the RS485 for the more interactive pieces. But with the RS485 port in close proximity there, it wouldn't be too difficult to add wires to the spare DB9 pins to connect to the RS485 port inside the case. Then the unit can do both if needed! Let's see what I can find by snooping the RS232 first and go from there :-)

OGB1952, 2023-03-25:
Could be MODBUS over RS232, perhaps?  

Would make sense for the inverter to use MODBUS internally and just have both the RS485 and RS232 connections to the same internal bus.  Could also be Solarman playing fast and loose when they state that the LSW-3 uses RS232.

OGB1952, 2023-03-25:
Just found another data sheet for the Solarman LSW-3 that has more info, although it is a bit confusing:

[image]

It mentions both RS485 and RS232, so I'm wondering if it's really just a duplicated RS485 port.  That would make some sense, with RS485 being multi-drop.  If we're REALLY lucky then it's really MODBUS over RS485, just like the other RS485 port.  If so, that would allow a direct connection for a Sofar2mqtt dongle, with the DB9 providing both power and data, with no mods.

I'd go out and test this right now, but my ME300SP is outside in a small shed, and it's tipping it down with rain here at the moment!

headcloudmonkey, 2023-03-25:
Hmmm, gave it a go and just CRC errors whichever way round I had the tx/rx

headcloudmonkey, 2023-03-25:
I'm suspecting the logger is just an esp8266 inside anyway, with the native tx/rx on it used to connect

headcloudmonkey, 2023-03-25:
[image]
[image]
[image]
[image]

I don't use the solarman app these days anyway, so I've disassembled the logger

headcloudmonkey, 2023-03-25:
tx/rx from the DB9 go to a Sipex 3232EE, which is an RS232 driver

https://www.sparkfun.com/datasheets/Components/General/sp3222_3232e.pdf

so we know it's just RS232 there then. I'm still trying to find my RS232 breakout cable...

headcloudmonkey, 2023-03-25:
https://fccid.io/2ACSV-HF-LPT230/User-Manual/Users-Manual-3552381.pdf

This is the module, provides wifi data transfer for serial devices.

OGB1952, 2023-03-25:
Great work!  Looks like the heavy lifting is done by this N76E003 Nuvoton microcontroller, with it's associated SPI flash memory chip:

[image]

https://www.nuvoton.com/products/microcontrollers/8bit-8051-mcus/low-pin-count-8051-series/n76e003/

The datasheet is here: https://www.nuvoton.com/export/resource-files/DS_N76E003_EN_Rev1.10.pdf

Seems to be a clone of the venerable 8051, so not the most capable device around.  Means it can't be doing much, I think, little more than serving up that very simple local webpage and handling the TCP/IP comms to the Solarman servers.  

Knowing that there is TTL level RS232 comms on that port gets us a bit closer to being able to make a variant of Sofar2mqtt that doesn't need an RS485 to RS232 converter.  I believe that the default baud rate for the Solarman LSW-3 is 9600, although the UART in the N76E003 can go up to 115,200.   The Sofar RS485 port runs at 9600 baud, so my best guess is that this RS232 port will be the same.

I may have a go tomorrow (weather permitting) to see if it's running MODBUS over RS232.  I think there is a fair chance it may be.  If so, then we could get really lucky and be able to just connect the software serial port of the ESP8266 to the DB9 Rx/Tx.  I have some DB9's around somewhere, and also have a spare breadboard Sofar2mqtt.  I will try and have a go at hooking something up, and, weather permitting, have a go with it tomorrow.