cmcgerty/Sofar2mqtt #29: upgraded to 2.0b1 but still CRC errors

javiermitchell, 2022-09-29:
I have upgraded to 2.0b1 and found it more stable - I think -
However, when I thoutgh everything was sorted I'm getting one of those dreaded "CRC-FAULT"errors 
So things are still not completely fixed, byt the looks of it.
The annoying thing is there doesn't seem to be quick solution: you just leave it as it is and it wil sort itself after it restarts the next day

In any case, I think this is a good effort

double1968, 2022-09-29:
since I have been using this adapter instead of the blue one I have not had any problems for now.
I'm on 2.0b1 too.
Also on github they recommend it

[https://www.amazon.it/gp/product/B096ZXXKCR/ref=ppx_od_dt_b_asin_image_s00?ie=UTF8&psc=1](url)

javiermitchell, 2022-09-29:
what do you mean?
this ONE is BLUE!

double1968, 2022-09-29:
[https://github.com/cmcgerty/Sofar2mqtt](url)

section How To Build

[image]

javiermitchell, 2022-09-29:
sorry, I misunderstood: you MEANT the blue BOARD, I thought you meant the blue CONNECTOR
so should be the red BOARD
Thanks

frogsuk, 2022-09-30:
> since using this adapter instead of the blue one I have had no more problems.
> 
> I'm on 2.0b1 too.
> 
> Also on github they recommend it
> 
> 
> 
> `https://www.amazon.it/gp/product/B096ZXXKCR/ref=ppx_od_dt_b_asin_image_s00?ie=UTF8&psc=1`

This is not the MAX3485 Chipped board as quoted in GitHub. It is the MAX13487E board and there is a difference.
Please see attached data sheets.
Always check the main chip in the board for it's number as not always described correctly

https://datasheets.maximintegrated.com/en/ds/MAX13487E-MAX13488E.pdf

https://datasheets.maximintegrated.com/en/ds/MAX3483E-MAX3491E.pdf

closed 2022-11-19 by cmcgerty