cmcgerty/Sofar2mqtt #40: node red flow does not work

Ernest-burgess, 2022-11-10:
hi the node red flow funtion node  does not seem to act on any of the payload values 
i have a debug node on what goes in  and one on the output  and they both show the same data it does not seem to be formating the data.  
ie the running state  shows as value 4 in the input   and still as value 4 in the output   the globle variable holds the string "Unknown"
so it seems to be droping right through the case statment  .
i have tried the funtion node that was uploaded a few days ago with revision 2 of the code  but this still does the same 
does any one have this working ? 
regards bill

cmcgerty, 2022-11-10:
Hi Bill,

The function works fine. Sorry if you are struggling to get it working in your system, I know it can be a pain. Could you ask your question in the discussion area?

[image]

Ernest-burgess, 2022-11-10:
Hi thanks for the prompt reply
That was what I was expecting to see  sorry I am new to this  this is a picture of what I am getting I wonder if it could be something in the set up of node red ? [cid:42225D64-80A4-4570-B874-E72D880D1839]

Regards. Bill

[Format]<https://user-images.githubusercontent.com/13269549/201046975-acdda849-5bef-4186-81d8-cb960e3586d9.png>

davies8172, 2022-11-10:
Check out this file to get you started. It is the original Sofar2MQTT flow for Node Red which works perfectly. It outputs to MQTT but you can change that to where-ever you like. :)
https://www.facebook.com/groups/2477195449252168/permalink/2570410946597284

Ernest-burgess, 2022-11-10:
Fixed it. Spent all last night on this. And your msg. Just made me think about the output from the Mqtt conection node. Set it to a parsed JSON object. Now works fine

Thanks

Regards. Bill