cmcgerty/Sofar2mqtt #45: Support for HYD 3-6K-EP

wardwygaerts, 2022-11-23:
If I'm correct there's only support for the HYD ES inverters, but not for the EP ones. These have other registries.
Are there plans to add this support?

cliffdude, 2023-02-08:
If someone could find the register value map it would certainly be possible. I may need to upgrade my ES inverter to a bigger EP model (better EPS mode and higher batt charge current) so I am interested.

lugio, 2023-02-11:
Hello, still no support planned ?
 thank you

cmcgerty, 2023-02-11:
Not at the moment, no. Sorry @lugio

lugio, 2023-02-11:
Thks for you response
😞

lugio, 2023-02-11:
this project manages to recover the information via rs485, it is possible to based on this one for the adapted to yours?
[https://gitlab.com/rjpearce/sofar2mqtt](url)

cliffdude, 2023-02-11:
@luigii

I am currently upgrading my solar install and may swap my ES inverter with
an EP model with higher power.
If that does happen I will attempt to add support for the EP as I would
like to keep my ESP device to be able to control the inverter.
A long shot but check back here in future, you never know :)

lugio, 2023-02-12:
I'm crossing my fingers then :)
 Is that what you're looking for? :

[esphome_sofarsolar_hydxxxxep_modbus.txt](https://github.com/cmcgerty/Sofar2mqtt/files/10716240/esphome_sofarsolar_hydxxxxep_modbus.txt)

cliffdude, 2023-02-12:
Thatis very useful, esphome is a great platform f9r anyone using home
assistant. I use it for many other sensors as the response time is way
better than mqtt.
Have you tried this out?

lugio, 2023-02-12:
Yes I am currently sending the data to Home Assistant
RS485 > USB > vm linux > mqqt

Thanks to the project:
https://gitlab.com/rjpearce/sofar2mqtt/-/blob/main/README.md

It works fine, but read only.

wardwygaerts, 2023-02-12:
I connected my EP to a RS485 to ETH server (Waveshare), which sends all data via mqtt. Is working fine!

rjpearce, 2023-02-20:
@cmcgerty I can probably change the code to support EP and send you an MR but I don't want to make your code complicated.

How would you feel about pulling out the data (eg. registry addresses) to a separate file per model i.e a JSON file or at least a .h file. For example: https://gitlab.com/rjpearce/sofar2mqtt/-/blob/main/sofar-hyd-ep.json

Splitting the data from the code makes the code easier to maintain and adding support for new models is as easy as adding a file. 

I also don't have a way to test but it looks like @lugio is up for that.

lugio, 2023-02-26:
Work base ? 
[(https://github.com/cmcgerty/Sofar2mqtt/discussions/57)]

joostvdveen, 2023-03-17:
I am looking for the HYD 3-6K-EP registry addresses to send Charge and Discharge command to the inverter. For me that is the most important functionality i need. anyone?

rjpearce, 2023-03-17:
> I am looking for the HYD 3-6K-EP registry addresses to send Charge and Discharge command to the inverter. For me that is the most important functionality i need. anyone?

https://www.facebook.com/groups/2477195449252168/permalink/3047428435562197
"0x1187" also known as 4487

joostvdveen, 2023-03-18:
> > I am looking for the HYD 3-6K-EP registry addresses to send Charge and Discharge command to the inverter. For me that is the most important functionality i need. anyone?
> 
> https://www.facebook.com/groups/2477195449252168/permalink/3047428435562197 "0x1187" also known as 4487

Sorry i don't have Facebook
[image]

rjpearce, 2023-03-19:
[Remote Control Description for SOFARSOLAR Hybird Inverter HYD (3).pdf](https://github.com/cmcgerty/Sofar2mqtt/files/11010905/Remote.Control.Description.for.SOFARSOLAR.Hybird.Inverter.HYD.3.pdf)

Loudo42, 2023-03-19:
Hi,

For those who want to have the good register and config, you can have a look to my reply here.
[https://github.com/cmcgerty/Sofar2mqtt/discussions/26#discussioncomment-4242730]

I still have to set up the remote control.
I had a look on the python code of the hyp-ep-remote-control.py of Andre Wagner in the facebook group, but I do not understand all the code to write the registers and I do not know how to adapt it in Sofar2mqtt.

I was thinking a easiest way to remote it.
The idea is to change only 1 register to switch between self use mode and time-of-use-mode.
In the inverter, we have to set up 1 rule of time of use mode, with a charge at 0W all the time, so it will correspond to standby mode.
Here is the code to change in Sofar2mqtt:
First for HYP-EP, I think we have to set up 
#define SOFAR_FN_AUTO	0x4368

In the void sendData() section:

if(cmd == "standby")
	{
		sendPassiveCmd(SOFAR_SLAVE_ID, SOFAR_FN_AUTO, 1, "standby")
//1 is Time-of-use mode, so set a enabled rule with charge 0W 
		Serial.println("standby");
		}
	}
	else if(cmd == "auto")
	{
		sendPassiveCmd(SOFAR_SLAVE_ID, SOFAR_FN_AUTO, 0, "auto")
		Serial.println("auto");
	}

Could you please let me know if it is ok and if I can try it.

Thank you in advance.

rjpearce, 2023-03-19:
Why not use Passive Mode (3) ? 
You can then specify the charge/discharge rate using register 0x1187. 

The code will need to be updated as SOFAR_FN_CHARGE and SOFAR_FN_DISCHARGE would be the same register but with a positive or negative value (negative=discharge, positive=charge). 

I've just incorporated Andre's code to bring write support to the [Raspberry Pi version of sofar2mqtt](https://gitlab.com/rjpearce/sofar2mqtt/-/tree/write_support/).

joostvdveen, 2023-03-20:
hi all, I just tested a new version https://github.com/IgorYbema/Sofar2mqtt/tree/mod of https://github.com/IgorYbema/Sofar2mqtt (a fork of https://github.com/cmcgerty/Sofar2mqtt) on my Solar HYD EP and the Charge and Discharge commands from home assistant work in passive mode.

wardwygaerts, 2023-03-20:
> hi all, I just tested a new version https://github.com/IgorYbema/Sofar2mqtt/tree/mod

Can I just flash the bin file, or do I need to build the code?

Loudo42, 2023-04-18:
Hi,

I have tried the IgorYbema fork code for my HYD6000EP. To make it working, I add to change a little the code.
But it misses the command to set it up in automode.
In the sofar description, it is said to write the register 4368 with 01 10 11 10 00 01 02 00 00 A5 C1.
Does somebody know  how to implement it?

Moreover, I can only send 23 values with the sendmqtt command. When I add 1 line more in the mqtt_status_register, nothing is sent.
What is the issue, how can I solve it?

Thank you very much in advance.