cmcgerty/Sofar2mqtt #51: HYD 6000-ZSS HP receiving mqtt with zero on all values

megamarco833, 2022-12-28:
hello, i have an inverter model HYD 6000-ZSS HP and i flashed a wemoss D1 mini 
```
//#define INVERTER_ME3000				// Uncomment for ME3000
#define INVERTER_HYBRID			// Uncomment for Hybrid
```

i'm using a TTL - RS485 generic (i'm waiting that MAX3485 will arrive)
https://ae01.alicdn.com/kf/H5299716877f943ae80442061597977df9.jpg
i connect
TTL-RS485  to  Wemos
 RX to D6
 TX to D7
 VCC to 3v3
 GND to GND

i see on oled: "standby" online
id i subscrive to Sofar2mqtt/state i see
`{"running_state":0,"grid_voltage":0,"grid_current":0,"grid_freq":0,"grid_power":0,"battery_power":0,"battery_voltage":0,"battery_current":0,"systemIO_power":0,"batterySOC":0,"battery_temp":0,"battery_cycles":0,"consumption":0,"solarPV":0,"solarPVAmps":0,"today_generation":0,"Solarpv1":0,"inverter_temp":0,"inverter_HStemp":0}`

so all vale are zero.

i have this plug and i connect my TTL RS485 to pin5 and pin6 of this plug (that's the only pin that display the message above)
and when i connect these two pins at main plug inside inverter display i see RS485 symbol in upper left corner
[image]
[image]

megamarco833, 2022-12-28:
i just notice that on OLED display on wemoss is showed:
```
sofar2mqtt
 online
 standby 
```

and after sometime just for a less then a second appear:
```
sofa2mqtt
 online
 RS485
 ERROR
```
then immediately the first message
[image]
[image]

https://user-images.githubusercontent.com/44502572/209865835-7e3bb270-5fda-476c-bf58-a841abf14b65.mp4

i also use arduino serial monitor and this is the log:

```
怘⸮ ⸮⸮⸮⸮⸮f⸮⸮ ~⸮⸮⸮⸮~  ~⸮⸮
WiFi connected - ESP IP address: 192.168.0.29
Attempting MQTT connection...connected
Set start up mode: Auto
Timeout waiting for RS485 response.
Response too short
Send heartbeat
Timeout waiting for RS485 response.
Response too short
Bad heartbeat -2
Get runstate: Timeout waiting for RS485 response.
Response too short
Response too short
Timeout waiting for RS485 response.
Response too short
Timeout waiting for RS485 response.
Response too short
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Send heartbeat
Timeout waiting for RS485 response.
Response too short
Bad heartbeat -2
Get runstate: Timeout waiting for RS485 response.
0
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Attempting MQTT connection...connected
Send heartbeat
Timeout waiting for RS485 response.
Response too short
Bad heartbeat -2
Get runstate: Timeout waiting for RS485 response.
0
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Send heartbeat
Timeout waiting for RS485 response.
Response too short
Bad heartbeat -2
Get runstate: Timeout waiting for RS485 response.
0
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
Timeout waiting for RS485 response.
```

cliffdude, 2022-12-29:
Hi,

Do you have the pinout for that connector? It is possible that only *one*
of the RS485 wires is in the correct pin, sometimes that will "trick" the
inverter into thinking that an RS485 is in fact connected and will show the
symbol in the display. The RS485 symbol will disappear after about 30
secs... It has happened to me :)
I cannot help you with the pinout as I have a different model.

megamarco833, 2022-12-29:
> Hi, Do you have the pinout for that connector? It is possible that only *one* of the RS485 wires is in the correct pin, sometimes that will "trick" the inverter into thinking that an RS485 is in [...]

hello, thanks for your reply.
finally i talked with customer care and they told me that the protocol is changed since 2021 with new generation of inverters (V3) and now all inverters mono phase and 3 pahses have the new common protocol RS485 modbus with new addresses.
i find the new adresses and i try to chenge your code with new one and i receive the right response, but the numbers aren't diplay correctly.
example: grid_freq":5001  should be 50,01

this is the json where i just try to change some addresses with new ones
`{"running_state":0,"grid_voltage":0,"grid_current":0,"grid_freq":5001,"grid_power":5001,"battery_power":65534,"battery_voltage":491,"battery_current":65466,"systemIO_power":65466,"batterySOC":27,"battery_temp":20,"battery_cycles":0,"consumption":0,"solarPV":0,"solarPVAmps":0,"today_generation":0,"Solarpv1":0,"Solarpv2":0,"today_consumption":0,"inverter_temp":0,"inverter_HStemp":0}`

i see in documentation that new now they are using 16bit and not 8bit could be that the reason?

this below is a configuration for HA that i find:

```
 - name: "Frequenza di rete" #"ongrid_frequency"
    slave: 1 
    address: 0x0484
    data_type: int16 
    device_class: frequency 
    unit_of_measurement: "Hz" 
    scale: 0.01
    scan_interval: 10
    precision: 2
```

do you like to add the support also for the new generation inverter? :)

double1968, 2022-12-29:
> hello, thanks for your reply. finally i talked with customer care and they told me that the protocol is changed since 2021 with new generation of inverters (V3) and now all inverters mono phase and 3 phases (mine is mono phase) have the new common protocol RS485 modbus with new addresses. i find the new addresses and i try to change your code with new one and i receive the right response, but the numbers aren't display correctly. example: grid_freq":5001 should be 50,01
> 
> this is the json where i just try to change some addresses with new ones `{"running_state":0,"grid_voltage":0,"grid_current":0,"grid_freq":5001,"grid_power":5001,"battery_power":65534,"battery_voltage":491,"battery_current":65466,"systemIO_power":65466,"batterySOC":27,"battery_temp":20,"battery_cycles":0,"consumption":0,"solarPV":0,"solarPVAmps":0,"today_generation":0,"Solarpv1":0,"Solarpv2":0,"today_consumption":0,"inverter_temp":0,"inverter_HStemp":0}`
> 
> i see in documentation that new now they are using 16bit and not 8bit could be that the reason?
> 
> this below is a configuration for HA that i find:
> 
> ```
>  - name: "Frequenza di rete" #"ongrid_frequency"
>     slave: 1 
>     address: 0x0484
>     data_type: int16 
>     device_class: frequency 
>     unit_of_measurement: "Hz" 
>     scale: 0.01
>     scan_interval: 10
>     precision: 2

For some values it is sufficient to change the scale value.
For other values such as battery power it is necessary to create template sensors to transform the value. You can find examples here too
[Configuration](https://github.com/cmcgerty/Sofar2mqtt/blob/main/Home%20Assistant%20example/configuration.yaml)

megamarco833, 2022-12-29:
> For some values it is sufficient to change the scale value. For other values such as battery power it is necessary to create template sensors to transform the value. You can find examples here too [...]

i would like to have the data corrected display by the code (.ino) because i don't have HA, but i have domoticz
my idea was to change to source code to have the compatibility with new inverters that has new addresses and 16bit data (but i'm not sure if the wrong number format is due to 16bit vs 8bit)
example frequency that i receive is 50000 instead of 50,00

double1968, 2022-12-29:
no sorry i don't have the new inverter.

megamarco833, 2022-12-29:
> no sorry i don't have the new inverter.

ah ok, sorry :)

may i ask why you have this calculation that is need only for these parameters below?

  ```
  - name: "grid_power"  
      state_topic: "Sofar2mqtt/state"
      unit_of_measurement: "W"
      device_class: power
      state_class: measurement
      value_template: >-
        {% if value_json.grid_power > (65535 / 2) %}
        {{ (65535 - value_json.grid_power) * 10 }}
        {% else %}
        {{ -(value_json.grid_power) * 10 }}
        {% endif %} 
    
    - name: "inverter_battery_power"
      state_topic: "Sofar2mqtt/state"
      unit_of_measurement: "W"
      device_class: power
      state_class: measurement
      value_template: > 
        {% if value_json.battery_power > (65535 / 2) %}
        {{ -(65535 - value_json.battery_power) * 10 }}
        {% else %}
        {{ (value_json.battery_power) * 10 }}
        {% endif %} 
        
    - name: "inverter_system_power"
      state_topic: "Sofar2mqtt/state"
      unit_of_measurement: "W"
      device_class: power
      state_class: measurement
      value_template: > 
        {% if value_json.systemIO_power > (65535 / 2) %}
        {{ (65535 - value_json.systemIO_power) * 10 }}
        {% else %}
        {{ -(value_json.systemIO_power) * 10 }}
        {% endif %} 
    
    - name: "inverter_battery_discharge"
      state_topic: "Sofar2mqtt/state"
      unit_of_measurement: "W"
      device_class: power
      state_class: measurement
      value_template: > 
        {% if (value_json.battery_power > 32767) %}
        {{ (65535 - value_json.battery_power) * 10 }}
        {% else %}
        {{ 0 }}
        {% endif %} 
    
    - name: "inverter_battery_charge"
      state_topic: "Sofar2mqtt/state"
      unit_of_measurement: "W"
      device_class: power
      state_class: measurement
      value_template: > 
        {% if (value_json.battery_power < 32767) %}
        {{ (value_json.battery_power) * 10 }}
        {% else %}
        {{ 0 }}
        {% endif %} 
```

double1968, 2022-12-30:
power and other data are returned with values to be processed with that formula.

megamarco833, 2023-01-27:
@cmcgerty this is the new protocol V3:
https://docs.google.com/spreadsheets/u/0/d/1p6p7n8-V2jlFid3FPPk5L4Sn4-tOWscQwpkahq5jRbs/htmlview#
seams that the new inverter has a new v3 protocol with new addresses

joostvdveen, 2023-03-15:
hi @megamarco833 
I have a SOFAR HYD 5000-EP with software V030021

I encounter the same issue, no values in HA

what software version do you have?
And is the thing I need to do change the addresses to new ones?

megamarco833, 2023-03-15:
> 

hello, my inverter is different model than yours, anyhow i discovered that the modbus protocol is different for new inverters, it has a different addresses.
in my case this was the reason why i was not able to get any value

joostvdveen, 2023-03-15:
@megamarco833 

> hello, my inverter is different model than yours, anyhow i discovered that the modbus protocol is different for new inverters, it has a different addresses.
> in my case this was the reason why i was not able to get any value

So I do need to replace the used addresses in  the Solar2mqtt.ino file to new ones I have to lookup in the google spreadsheet??

```c++

// SoFar ME3000 Information Registers
#define SOFAR_REG_RUNSTATE	0x0200
#define SOFAR_REG_GRIDV		0x0206
#define SOFAR_REG_GRIDA		0x0207
#define SOFAR_REG_GRIDFREQ	0x0484
#define SOFAR_REG_BATTW		0x020d
#define SOFAR_REG_BATTV		0x020e
#define SOFAR_REG_BATTA		0x020f
#define SOFAR_REG_BATTSOC	0x0608
#define SOFAR_REG_BATTTEMP	0x0211
#define SOFAR_REG_GRIDW		0x0212
#define SOFAR_REG_LOADW		0x0213
#define SOFAR_REG_SYSIOW	0x0214
#define SOFAR_REG_PVW		0x0215
#define SOFAR_REG_PVDAY		0x0218
#define SOFAR_REG_EXPDAY	0x0219
#define SOFAR_REG_IMPDAY	0x021a
#define SOFAR_REG_LOADDAY	0x021b
#define SOFAR_REG_BATTCYC	0x022c
#define SOFAR_REG_PVA		0x0236
#define SOFAR_REG_INTTEMP	0x0238
#define SOFAR_REG_HSTEMP	0x0239
#define SOFAR_REG_PV1		0x0252
#define SOFAR_REG_PV2		0x0255
#define SOFAR_FN_STANDBY	0x0100
#define SOFAR_FN_DISCHARGE	0x0101
#define SOFAR_FN_CHARGE		0x0102
#define SOFAR_FN_AUTO		0x0103
```

joostvdveen, 2023-03-15:
I found another thread https://github.com/cmcgerty/Sofar2mqtt/discussions/57 I will ook into that

megamarco833, 2023-03-16:
....mine register are completely different:

```
//HYD 6000 ZSS HB -- V3 protocol
  #define SOFAR_REG_RUNSTATE  0x0404
  #define SOFAR_REG_GRIDV   0x048D
  #define SOFAR_REG_GRIDA   0x048E
  #define SOFAR_REG_GRIDFREQ  0x0484

  #define SOFAR_PV1_VOLT    0x0584
  #define SOFAR_PV1_CURR    0x0585
  #define SOFAR_PV1_POW     0x0586
  #define SOFAR_PV2_VOLT    0x0587
  #define SOFAR_PV2_CURR    0x0588
  #define SOFAR_PV2_POW     0x0589

  #define SOFAR_active_power     0x0485
  #define SOFAR_today_generation     0x0685
  #define SOFAR_total_generation     0x0687
  #define SOFAR_load_power_house     0x04AF  //AC on Grid output ==  ActivePower_Load_Sys
  #define SOFAR_load_power_pv_ext     0x04AE //ActivePower_PV_Ext
  #define SOFAR_load_power_out_tot      0x0485  //ActivePower_Output_Total
  #define SOFAR_load_power_pcc_tot      0x0488 //ActivePower_PCC_Total
  #define SOFAR_load_power_grid5      0x0506
  #define SOFAR_load_power_grid6      0x0507
  #define SOFAR_load_power_grid7      0x050A
  #define SOFAR_load_power_grid8      0x0524

  #define SOFAR_today_consumption     0x0689
  #define SOFAR_total_consumption     0x068B

  #define SOFAR_today_import     0x068D
  #define SOFAR_total_import     0x068F
  #define SOFAR_today_export     0x0691
  #define SOFAR_total_export     0x0693

  #define SOFAR_REG_BATTW   0x0606
  #define SOFAR_BATT_TEMP   0x0607
  #define SOFAR_BATT_CURR   0x0605
  #define SOFAR_BATT_VOLT   0x0604
  #define SOFAR_BATT_TDY_DISC   0x0699
  #define SOFAR_BATT_TDY_CHARG   0x0695
  #define SOFAR_BATT_SOC 0x0608    // SOC state of charge battery1== stato di carica in %
  #define SOFAR_BATT_SOC2 0x060F    // SOC state of charge battery2 == stato di carica in %
  #define SOFAR_BATT_HEALT 0x0609  //battery_soh
  #define SOFAR_BATT_CYCLES 0x060A
  #define SOFAR_REG_BATTW   0x0606
  #define SOFAR_REG_BATTV   0x0604
  #define SOFAR_REG_BATTA   0x0605
```