cmcgerty/Sofar2mqtt #53: Can Sofar2mqtt be RS485 slave?

Simoverdex, 2023-01-14:
Hi folks, in the last few days I've added a wallbox for charging electric cars, which communicates with my inverter in RS485. I tried to connect my Sofar2mqtt together with the walbox, but being two Modbus Masters, I often receive incorrect values. Is there a possibility to make our Sofar2mqtt RS485 slave, so it would become a data "Sniffer"?

phamels, 2023-02-12:
Also interested to find out.
I noticed there is a slave port on the Sofar controller. Would it automatically become a slave without the ability to send data to it (just querying) or how would that work?
Anyone have any idea or more info on the slave port?

Simoverdex, 2023-02-12:
> 

So, in a modbus network the master (in this case the logger) makes the request, while the slave (inverter) sends the data. In my situation I have two masters requesting data from a slave, and this situation generates errors.

phamels, 2023-02-12:
I have a similar issue where I want to use sofar2mqtt as a master, but a secondary device as a master as well.
Just confused with the fact that the sofar controller ME3000SP in my case, has a RS485 master and slave port.
So I guess the way around this would be if we could for example modify sofar2mqtt to act as a master, read the data and replay it to the second master? Just thinking out loud here.