cmcgerty/Sofar2mqtt #56: wrong order of if statement?

IgorYbema, 2023-02-22:
Should this be the other way around? 
https://github.com/cmcgerty/Sofar2mqtt/blob/4f061801fb2c7e3699ccd78cc1b99fed54aead0d/Sofar2mqtt/Sofar2mqtt.ino#L459-L462

cmcgerty, 2023-02-22:
Quite possibly. It certainly looks like p would only be set if readSingleReg fails. Do you have one connected with a serial console open? Does it constantly log “modbus error” when in battery save mode? I’ll take a look when I get a chance.

IgorYbema, 2023-02-22:
Not using batterysave but I just noticed it while modifying the code a bit to suit my own situation

david9600, 2025-05-27:
I know this thread dates back a couple of years, but as I had a similar thought when reading the code, I looked into it, and can confirm that the logic is OK as is.

The readSingleReg() function passes on the return code from sendModbus(), which in turn passes on the return code from listen(). That function listen() returns an errorLevel which is zero for a valid frame received, or non-zero otherwise.

So, if(readSingleReg) is equivalent to "if a bad frame is received", and if(!readSingleReg) is "if a valid frame is received", which is what's needed.  A bit confusing yes, but it seems OK, and is working in my ESP, which is the main thing :P)

closed 2025-05-27 by cmcgerty