cmcgerty/Sofar2mqtt #59: Compiling Errors with latest Arduino IDE 2.0.4

championc, 2023-03-12:
Hi there

When I take the code from the project, I get the following errors when compiling it - although it does compile, but can this reported errors be fixed ?

C:\Users\corma\Downloads\Sofar2mqtt\Sofar2mqtt.ino: In function 'int listen(modbusResponse*)':
C:\Users\corma\Downloads\Sofar2mqtt\Sofar2mqtt.ino:644:24: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  644 |   resp->errorMessage = "Response too short";
      |                        ^~~~~~~~~~~~~~~~~~~~
C:\Users\corma\Downloads\Sofar2mqtt\Sofar2mqtt.ino:649:24: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  649 |   resp->errorMessage = "Valid data frame";
      |                        ^~~~~~~~~~~~~~~~~~
C:\Users\corma\Downloads\Sofar2mqtt\Sofar2mqtt.ino:654:24: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  654 |   resp->errorMessage = "Error: invalid data frame";
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\corma\Documents\Arduino\libraries\Adafruit_SSD1306_Wemos_Mini_OLED\Adafruit_SSD1306.cpp: In member function 'void Adafruit_SSD1306::drawFastHLineInternal(int16_t, int16_t, int16_t, uint16_t)':
c:\Users\corma\Documents\Arduino\libraries\Adafruit_SSD1306_Wemos_Mini_OLED\Adafruit_SSD1306.cpp:612:21: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  612 |   register uint8_t *pBuf = buffer;
      |                     ^~~~
c:\Users\corma\Documents\Arduino\libraries\Adafruit_SSD1306_Wemos_Mini_OLED\Adafruit_SSD1306.cpp:618:20: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  618 |   register uint8_t mask = 1 << (y&7);
      |                    ^~~~
c:\Users\corma\Documents\Arduino\libraries\Adafruit_SSD1306_Wemos_Mini_OLED\Adafruit_SSD1306.cpp: In member function 'void Adafruit_SSD1306::drawFastVLineInternal(int16_t, int16_t, int16_t, uint16_t)':
c:\Users\corma\Documents\Arduino\libraries\Adafruit_SSD1306_Wemos_Mini_OLED\Adafruit_SSD1306.cpp:686:20: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  686 |   register uint8_t y = __y;
      |                    ^
c:\Users\corma\Documents\Arduino\libraries\Adafruit_SSD1306_Wemos_Mini_OLED\Adafruit_SSD1306.cpp:687:20: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  687 |   register uint8_t h = __h;
      |                    ^
c:\Users\corma\Documents\Arduino\libraries\Adafruit_SSD1306_Wemos_Mini_OLED\Adafruit_SSD1306.cpp:691:21: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  691 |   register uint8_t *pBuf = buffer;
      |                     ^~~~
c:\Users\corma\Documents\Arduino\libraries\Adafruit_SSD1306_Wemos_Mini_OLED\Adafruit_SSD1306.cpp:698:20: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  698 |   register uint8_t mod = (y&7);
      |                    ^~~
c:\Users\corma\Documents\Arduino\libraries\Adafruit_SSD1306_Wemos_Mini_OLED\Adafruit_SSD1306.cpp:706:22: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  706 |     register uint8_t mask = premask[mod];
      |                      ^~~~
c:\Users\corma\Documents\Arduino\libraries\Adafruit_SSD1306_Wemos_Mini_OLED\Adafruit_SSD1306.cpp:744:24: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  744 |       register uint8_t val = (color == WHITE) ? 255 : 0;
      |                        ^~~
c:\Users\corma\Documents\Arduino\libraries\Adafruit_SSD1306_Wemos_Mini_OLED\Adafruit_SSD1306.cpp:766:22: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  766 |     register uint8_t mask = postmask[mod];
      |                      ^~~~

joostvdveen, 2023-03-16:
I have the same issue

OGB1952, 2023-03-21:
Same here, too.  The previous version of Sofar2mqtt was fine, seems to compile without errors.  I'm only getting the compile warnings in the CRC section, though, but that may be because I'm using a different display.

JonathanJ71, 2023-05-04:
Is somebody have a compilable version for ide 2?

nejib1, 2023-08-18:
Just comment these lines... no need of them

davergu, 2024-02-25:
Has this been resolved? I have tried various versions of Arduino IDE and Sofar2mqtt and none have worked. I've not attempted older libraries as the permutations are legion. Does anyone know the required permutation of everything?!

rbrooklyn, 2025-05-31:
It compiles fine, you do need to ensure you have installed specific versions of the libraries and not the latest versions which have introduced breaking changes.

I have compiled this project today successfully using Arduino IDE 2.3.6 and these libraries and versions:
Adafruit BusIO 1.14.1
Adafruit GFX Library 1.12.1
Adafruit SSD1306 Wemos Mini OLED 1.1.2
PubSubClient 2.8
And these boards and versions:
Arduino AVR Boards 1.8.6
esp8266 3.0.2

david9600, 2025-05-31:
My experience with current libraries was that the compilation completed, but with 3 warnings in the listen() function relating to the conversion of string to a char pointer.
While the warnings didn't prevent loading the code, I eventually researched the errors with the aim of clean compilation, and was able to modify the code to that below, which converts the string before insertion into the array.  In my IDE at least, it now runs cleanly!

	String str;
	if(inFrameSize < 5)
	{
		resp->errorLevel = 2;
		str = "Response too short";
	}
	else if(checkCRC(inFrame, inFrameSize))
	{
		resp->errorLevel = 0;
		str = "Valid data frame";
	}
	else
	{
		resp->errorLevel = 1;
		str = "Error: invalid data frame";
	}
	//convert string to char*
	std::vector<char> cstr(str.begin(), str.end());
	resp->errorMessage = cstr.data();
        
        return -resp->errorLevel;