cmcgerty/Sofar2mqtt #70: Recovery of product power only

stephcache, 2023-10-28:
Hello, I have seen your code and I cannot modify it in order to recover only the power produced in order to turn on my water heater, heating.

[https://forum.arduino.cc/t/rs485-et-sofar-solar/1180679](url)

I would really like some help to retrieve this information only, I don't need a home assistant

cliffdude, 2023-10-29:
You can try my fork, it is of an older version however you can get the data
you need in JSON format.
https://github.com/cliffdude/SofarCtrl

You will need to multiply the solarPV by 10 and ignore the rest of the data
in the JSON string.

Good luck

[image: image.png]

stephcache, 2023-10-29:
Your link looks a lot like the other one. I don't need oled, nor mqtt, nor json and I use an esp32 not an esp8266.

In addition <SoftwareSerial.h> is not visible, not compatible with esp32.

I am looking for the power produced by the instantaneous panels.

I can't find any information to just read the registers that come in HYD 6000-EP

cliffdude, 2023-10-29:
If you want to use an ESP32 you will have to modify the code yourself, the
info for the registers is in the code on Colin's repo, my fork is not
compatible with the EP version of the inverter  ( I have the older ES).
I would suggest creating a repo targeted at the ESP32 board, look through
the code and you will find the section that communicates with the inverter,
that is all you need. Extract it and use it in your own code.

stephcache, 2023-10-29:
I go through them, and I can't find the part that communicates with the inverter (I don't understand would be more accurate). That’s why I’m asking the question.

cliffdude, 2023-10-29:
Line 543 to 700 contains most of the code, you will need to read it to
understand what is going on.

https://github.com/cmcgerty/Sofar2mqtt/blob/main/Sofar2mqtt/Sofar2mqtt.ino

Check out the pull requests as there is more info on the registers for the
EP.
 I did not write this code and have little experience with modbus so I
cannot help you more.